micro-ROS/micro_ros_setup #677: What is the correct way to create a message variable

elgarbe, 2023-11-29:
Hi, I was looking for information abut the proper way of create a message in the MCU firmware.

I saw code like this.

```
geometry_msgs__msg__Vector3Stamped__Sequence * speed_message;

// Vector Stamped 3
speed_message = geometry_msgs__msg__Vector3Stamped__Sequence__create(4);
geometry_msgs__msg__Vector3Stamped__Sequence__init(speed_message, 4);

speed_message->data[0].header.frame_id.data =  (char*)malloc(100 * sizeof(char));

char string3[] = "/robot_rpm";
memcpy(speed_message->data[0].header.frame_id.data, string3, strlen(string3) + 1);
speed_message->data[0].header.frame_id.size = strlen(speed_message->data[0].header.frame_id.data);
speed_message->data[0].header.frame_id.capacity = 100;
```

and wondering were to find information of the differents functions used here, what the _Sequence_create is, why it has a 4 as parameter, the same for _Sequence_init

Thank!

pablogs9, 2023-11-30:
Here: https://docs.vulcanexus.org/en/latest/rst/tutorials/micro/memory_management/memory_management.html#message-memory

elgarbe, 2023-11-30:
> Here: https://docs.vulcanexus.org/en/latest/rst/tutorials/micro/memory_management/memory_management.html#message-memory

Thank! I now know what a Sequence type is, I didn't know it, I'm looking for compund message type like vecto3stamped.
Following your link I found [this one](https://micro.ros.org/docs/tutorials/advanced/handling_type_memory/) were the micro_ros_utilities_create_message_memory is explained. I will try to use it.

Thank!

pablogs9, 2023-11-30:
solved here: https://github.com/micro-ROS/micro_ros_stm32cubemx_utils/issues/132

closed 2023-11-30 by pablogs9