micro-ROS/micro_ros_setup #705: Micro-ROS Agent Build Fail - DDS_CDR

n0sc3tipsum, 2024-06-08:
## Issue template

- Hardware description: Ubuntu 22.04, ESP32 DevkitC
- RTOS: No RTOS implemented
- Installation type: micro-ROS setup
- Version or commit hash:humble

#### Steps to reproduce the issue

1. Follow instructions in : https://micro.ros.org/docs/tutorials/core/first_application_linux/
2. build agent 

#### Expected behavior

Agent builds successfully

#### Actual behavior

For some reason, during building of the micro ros agent I get the following errors :  (note : I did try install micro_ros_setup a few times before but had to re-install it as I was facing some other issues. The build in this issue is from a newly setup microros_ws.)

```
ros2 run micro_ros_setup build_agent.sh

Building micro-ROS Agent

[0.544s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:

	'builtin_interfaces' is in: /home/n0sc3tipsum/microros/install/builtin_interfaces, /opt/ros/humble

	'micro_ros_msgs' is in: /home/n0sc3tipsum/microros/install/micro_ros_msgs

If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.

If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:

	--allow-overriding builtin_interfaces micro_ros_msgs

This may be promoted to an error in a future release of colcon-override-check.

Starting >>> builtin_interfaces

Starting >>> micro_ros_msgs

Finished <<< builtin_interfaces [4.51s]                                                                    

Finished <<< micro_ros_msgs [4.73s]                  

Starting >>> micro_ros_agent

--- stderr: micro_ros_agent                             

/home/n0sc3tipsum/microros/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_typesupport.cpp: In member function ‘virtual bool uros::agent::graph_manager::ParticipantEntitiesInfoTypeSupport::serialize(void*, eprosima::fastrtps::rtps::SerializedPayload_t*)’:

/home/n0sc3tipsum/microros/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_typesupport.cpp:57:33: error: ‘DDS_CDR’ is not a member of ‘eprosima::fastcdr::Cdr’

   57 |         eprosima::fastcdr::Cdr::DDS_CDR);

      |                                 ^~~~~~~

/home/n0sc3tipsum/microros/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_typesupport.cpp:64:54: error: ‘class eprosima::fastcdr::Cdr’ has no member named ‘getSerializedDataLength’; did you mean ‘get_serialized_data_length’?

   64 |         payload->length = static_cast<uint32_t>(scdr.getSerializedDataLength());

      |                                                      ^~~~~~~~~~~~~~~~~~~~~~~

      |                                                      get_serialized_data_length

/home/n0sc3tipsum/microros/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_typesupport.cpp: In member function ‘virtual bool uros::agent::graph_manager::ParticipantEntitiesInfoTypeSupport::deserialize(eprosima::fastrtps::rtps::SerializedPayload_t*, void*)’:

/home/n0sc3tipsum/microros/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_typesupport.cpp:80:33: error: ‘DDS_CDR’ is not a member of ‘eprosima::fastcdr::Cdr’

   80 |         eprosima::fastcdr::Cdr::DDS_CDR);

      |                                 ^~~~~~~

/home/n0sc3tipsum/microros/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_typesupport.cpp: In member function ‘virtual bool uros::agent::graph_manager::MicrorosGraphInfoTypeSupport::serialize(void*, eprosima::fastrtps::rtps::SerializedPayload_t*)’:

/home/n0sc3tipsum/microros/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_typesupport.cpp:149:33: error: ‘DDS_CDR’ is not a member of ‘eprosima::fastcdr::Cdr’

  149 |         eprosima::fastcdr::Cdr::DDS_CDR);

      |                                 ^~~~~~~

/home/n0sc3tipsum/microros/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_typesupport.cpp:156:54: error: ‘class eprosima::fastcdr::Cdr’ has no member named ‘getSerializedDataLength’; did you mean ‘get_serialized_data_length’?

  156 |         payload->length = static_cast<uint32_t>(scdr.getSerializedDataLength());

      |                                                      ^~~~~~~~~~~~~~~~~~~~~~~

      |                                                      get_serialized_data_length

/home/n0sc3tipsum/microros/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_typesupport.cpp: In member function ‘virtual bool uros::agent::graph_manager::MicrorosGraphInfoTypeSupport::deserialize(eprosima::fastrtps::rtps::SerializedPayload_t*, void*)’:

/home/n0sc3tipsum/microros/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_typesupport.cpp:172:33: error: ‘DDS_CDR’ is not a member of ‘eprosima::fastcdr::Cdr’

  172 |         eprosima::fastcdr::Cdr::DDS_CDR);

      |                                 ^~~~~~~

gmake[2]: *** [CMakeFiles/micro_ros_agent.dir/build.make:118: CMakeFiles/micro_ros_agent.dir/src/agent/graph_manager/graph_typesupport.cpp.o] Error 1

gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/micro_ros_agent.dir/all] Error 2

gmake: *** [Makefile:146: all] Error 2

---

Failed   <<< micro_ros_agent [2.35s, exited with code 2]

                                

Summary: 2 packages finished [7.56s]

  1 package failed: micro_ros_agent

  1 package had stderr output: micro_ros_agent

[ros2run]: Process exited with failure 2

```

#### Additional information

pablogs9, 2024-06-10:
Just following the instructions you mentioned: https://asciinema.org/a/ruiOV0pa9pybhgL152t8HyQfF

closed 2024-06-10 by pablogs9

n0sc3tipsum, 2024-06-10:
> Just following the instructions you mentioned: https://asciinema.org/a/ruiOV0pa9pybhgL152t8HyQfF

Yes I was able to get it working on a fresh ROS2 installation in docker, but on my actual laptop it doesn't seem to work. Would I have to just start with a  fresh ROS2 setup if I want it to work, or is there something I can clear/update, as I did try to setup micro-ros multiple times by deleting its' directory and re-starting the installation.

Thank you!

Muthulakshimi, 2024-10-03:
@n0sc3tipsum were you able to get any solution for working with ROS2 directly without fresh installation every time