micro-ROS/micro_ros_setup #711: error: ‘DDS_CDR’ is not a member of ‘eprosima::fastcdr::Cdr’

tuananh0606-tech, 2024-07-10:
## Issue template

-Hardware description: Dell G5
-RTOS: Linux
-Installation type: micro_ros_setup
-Version or commit hash: humble

#### Steps to reproduce the issue
I'm trying to install micro_ros_setup but when it comes to ros2 run micro_ros_setup build_agent.sh it gives me this error, I've tried many ways but it doesn't work, I've installed MicroXrceDDS before, is this the cause, if not, is there another way to fix it. Help me, please! Thanks.

#### Expected behavior
[image]
[image]

#### Actual behavior

#### Additional information

Tilgaard, 2024-07-15:
Hey [tuananh0606-tech](https://github.com/tuananh0606-tech)

Any progress? I have the same issue.

tuananh0606-tech, 2024-07-24:
> Hey [tuananh0606-tech](https://github.com/tuananh0606-tech)
> 
> Any progress? I have the same issue
Have you ever downloaded xcredds?

tuananh0606-tech, 2024-07-24:
@Tilgaard Have you ever downloaded xcredds?

NustAirworks2022, 2024-08-02:
yes I have downloaded XCREDDS and is facing the same issue

tuananh0606-tech, 2024-08-28:
@NustAirworks2022 have you solved it? I don't know how to completely remove XCRE-DDS

tuananh0606-tech, 2024-09-05:
@NustAirworks2022 Can you show me how you do it? thank you

NustAirworks2022, 2024-09-05:
Enter the following commands to fetch and build the micro XCRE-DDS agent from source:

`git clone https://github.com/eProsima/Micro-XRCE-DDS-Agent.git
cd Micro-XRCE-DDS-Agent
mkdir build
cd build
cmake ..
make
sudo make install
sudo ldconfig /usr/local/lib/`

start the agent to connect to simulator
`MicroXRCEAgent udp4 -p 8888`

Varchasvin10, 2024-10-03:
Im getting the same build error while trying to setup micro_ros in ros2 humble. Has any solution been found? Before the micro ros was working fine but when I downloaded and used XRCE-DDS for PX4 as @NustAirworks2022 mentioned micro ros stopped publishing messages and when i tried to reinstall it showed the same error as @tuananh0606-tech. 
[image]
I dont want to continue using XRCE DDS, is there a way to delete this, i tried deleting the folders but still doesnt work.

Varchasvin10, 2024-10-03:
> Im getting the same build error while trying to setup micro_ros in ros2 humble. Has any solution been found? Before the micro ros was working fine but when I downloaded and used XRCE-DDS for PX4 as @NustAirworks2022 mentioned micro ros stopped publishing messages and when i tried to reinstall it showed the same error as @tuananh0606-tech.
> 
> ![Screenshot from 2024-10-03 19-47-12](https://private-user-images.githubusercontent.com/117083612/373288980-62c743dc-2926-4b7a-a358-9931ad4d1b91.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc5NjcwMTUsIm5iZiI6MTcyNzk2NjcxNSwicGF0aCI6Ii8xMTcwODM2MTIvMzczMjg4OTgwLTYyYzc0M2RjLTI5MjYtNGI3YS1hMzU4LTk5MzFhZDRkMWI5MS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAwM1QxNDQ1MTVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00MTcwOTkxY2E0MDQwYTQ5YWE4ZTlmZmU4YWY3Y2I0NzNlMTRmZDQ2NWE0ZGRkZTU3MzYzYmIyMDdiN2VjZDBhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.Hw8xCW6Xk5kEWu6d5sBeOMLL5SGuIHL-k8W2RT8dVhE)
> 
> I dont want to continue using XRCE DDS, is there a way to delete this, i tried deleting the folders but still doesnt work.

Got it working just had to sudo rm -r the xrcedds packages in the root directory.

Karthik03B, 2024-10-09:
@Varchasvin10  I tried removing all the packages in the root directory, but still it doesn't work. It will help if you can give more details about how you did it and what are the packages that you installed and removed.
Thank you

Varchasvin10, 2024-10-09:
> @Varchasvin10 I tried removing all the packages in the root directory, but still it doesn't work. It will help if you can give more details about how you did it and what are the packages that you installed and removed. Thank you

Open up the terminal and try typing these,
`sudo rm -r microxrcedds_agent
sudo rm -r microxrcedds_client`

Ensure that you have manually deleted the MIcroXRCE-DDS packages from your home directory.

Let me know if this works, it worked for me.

Karthik03B, 2024-10-09:
@Varchasvin10 I manually deleted all the DDS packages. It didn't work for me.

Varchasvin10, 2024-10-09:
@Karthik03B DId you run the commands to remove it from your root directory?

Karthik03B, 2024-10-09:
yes I did. there were no such files in the root.

Varchasvin10, 2024-10-09:
> yes I did. there were no such files in the root.
> 

Delete your current microros workspace and Retry the microros setup and installation guidelines, you might have built the workspace before and that might not be allowing you to build it again as it might be saved with the previous setup.

Karthik03B, 2024-10-10:
@Varchasvin10 I tried that also. Before this i installed XRCE DDS in the root , I think i installed it Globally. It's just my assumption.