micro-ROS/micro_ros_setup #764: micro_ros_agent installation issue on ROS 2 Humble Ubuntu 22.04

TannerGilbert, 2025-01-13:
## Issue template

- Hardware description: PC, Ubuntu 22.04
- RTOS: Ardupilot SITL
- Installation type: micro_ros_setup
- Version or commit hash: humble

#### Steps to reproduce the issue

Follow [First micro-ROS Application on Linux](https://micro.ros.org/docs/tutorials/core/first_application_linux/). When creating and building the micro-ROS agent

```
ros2 run micro_ros_setup create_agent_ws.sh
ros2 run micro_ros_setup build_agent.sh
```
I run into the following error:
```
stderr: micro_ros_agent
CMake Error at /opt/ros/humble/share/fastrtps/cmake/fastrtps-dynamic-targets.cmake:37 (message):
Some (but not all) targets in this export set were already defined.

Targets Defined: eProsima_atomic

Targets not yet defined: fastrtps

Call Stack (most recent call first):
/opt/ros/humble/share/fastrtps/cmake/fastrtps-config.cmake:60 (include)
CMakeLists.txt:37 (find_package)
```

#### Expected behavior

micro-ROS agent installation works as expected and I can use it for working with Ardupilot.

#### Actual behavior

```
stderr: micro_ros_agent
CMake Error at /opt/ros/humble/share/fastrtps/cmake/fastrtps-dynamic-targets.cmake:37 (message):
Some (but not all) targets in this export set were already defined.

Targets Defined: eProsima_atomic

Targets not yet defined: fastrtps

Call Stack (most recent call first):
/opt/ros/humble/share/fastrtps/cmake/fastrtps-config.cmake:60 (include)
CMakeLists.txt:37 (find_package)
```

#### Additional information

Installing Fast-DDS from source didn't fix the issue for me either

seifseghiri, 2025-01-14:
Hi Tanner,
Any solution yet for fixing this?
Thanks in advance.

starrtriaxle82, 2025-01-27:
Did you figure out this issue? I'm experiencing the same error.

Targets Defined: eProsima_atomic

Targets not yet defined: fastrtps

TannerGilbert, 2025-01-28:
No I haven't found a solution to this yet.

Ericc99, 2025-02-13:
I got the same problem as well ... hope someone can figure it out soon ...
Before that, may i suggest you use a docker image instead with the following command 

```
sudo docker run -it --rm --net=host microros/micro-ros-agent:humble udp4 --port 8888 -v6
```

this works for me at the moment.

but watch out for your ufw status, or else all signals will be blocked lol.

p8410077, 2025-03-19:
@TannerGilbert did you resolve the problem ? I ran into the same issue.

TannerGilbert, 2025-03-19:
No unfortunately I didn't solve this yet. The error only occurs on one of my devices though so currently I just run on a different one. The solution proposed by @Ericc99 should probably also solve the issue but I haven't tested it myself yet.

seifseghiri, 2025-07-18:
It was a long time ago, but I think I found the issue. Which was that I accidentally installed a wrong version of eProsima and I believe it was installed in /opt/ or snap. Make sure no eProsima version is installed, outside of the one coming from installing ROS.

lijin166, 2025-07-24:
I'm facing the same problem again today. While it seems to be fixed now, I'm unclear about the exact solution. I performed just two specific actions: reinstalling the ros-jazzy-rmw-fastrtps-* package and rebuilding the 'micro-xrce-dds' software.
`sudo apt reinstall ros-jazzy-rmw-fastrtps-*`
`git clone https://github.com/eProsima/Micro-XRCE-DDS-Client.git` and install it. 
I hope this could do helpful.

Dworf2882, 2026-02-28:
Ubuntu 22.04
ros2 humble

4 a.m. the whole problem was that I got a few more versions of fastdds from somewhere, with which I knocked down Micro-XRCE-DDS.After cleaning, everything worked, thank God

```bash
alex@alex:~/ardu_ws$ ls /usr/local/lib | grep fast
ls /usr/local/lib | grep fastrtps
ls /usr/local/include | grep fast
libfastcdr.so
libfastcdr.so.2.2.7
libfastdds.so
libfastdds.so.3.5.0.0
fastcdr
fastdds
alex@alex:~/ardu_ws$ sudo rm -f /usr/local/lib/libfastcdr.so*
sudo rm -f /usr/local/lib/libfastdds.so*
sudo rm -rf /usr/local/include/fastcdr
sudo rm -rf /usr/local/include/fastdds
sudo ldconfig
[sudo] password for alex:
alex@alex:~/ardu_ws$ ls /usr/local/include | grep fast
alex@alex:~/ardu_ws$ rm -rf build install log
alex@alex:~/ardu_ws$ echo $RMW_IMPLEMENTATION
rmw_cyclonedds_cpp
alex@alex:~/ardu_ws$ colcon build --packages-up-to ardupilot_dds_tests
Starting >>> ardupilot_msgs
Starting >>> micro_ros_agent
Finished <<< ardupilot_msgs [6.60s]
--- stderr: micro_ros_agent
Cloning into 'xrceagent'...
HEAD is now at 57d0862 Release v2.4.2
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (tinyxml2)
  does not match the name of the calling package (TinyXML2).  This can lead
  to problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/modules/FindTinyXML2.cmake:40 (find_package_handle_standard_args)
  /opt/ros/humble/share/fastrtps/cmake/fastrtps-config.cmake:51 (find_package)
  CMakeLists.txt:153 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at /usr/local/lib/cmake/fastcdr/fastcdr-shared-targets.cmake:75 (message):
  The imported target "fastcdr" references the file

     "/usr/local/lib/libfastcdr.so.2.2.7"

  but this file does not exist.  Possible reasons include:

  * The file was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and contained

     "/usr/local/lib/cmake/fastcdr/fastcdr-shared-targets.cmake"

  but not all the files it references.

Call Stack (most recent call first):
  /usr/local/lib/cmake/fastcdr/fastcdr-config.cmake:80 (include)
  /usr/local/lib/cmake/fastcdr/fastcdr-config.cmake:101 (fastcdr_load_targets)
  /usr/local/share/microxrcedds_agent/cmake/microxrcedds_agentConfig.cmake:53 (find_package)
  CMakeLists.txt:34 (find_package)

gmake[2]: *** [CMakeFiles/micro_ros_agent.dir/build.make:94: micro_ros_agent-prefix/src/micro_ros_agent-stamp/micro_ros_agent-configure] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:111: CMakeFiles/micro_ros_agent.dir/all] Error 2
gmake: *** [Makefile:91: all] Error 2
---
[... 161 lines cut ...]
        where the options are:
                -help: shows this help
                -version: shows the current version of eProsima Micro XRCE-DDS Gen.
                -example: Generates an example.
                -replace: replaces existing generated files.
                -ppDisable: disables the preprocessor.
                -ppPath: specifies the preprocessor path.
                -I <path>: add directory to preprocessor include paths.
                -d <path>: sets an output directory for generated files.
                -t <temp dir>: sets a specific directory as a temporary directory.
                -cs: IDL grammar apply case sensitive matching.
        and the supported input files are:
        * IDL files.
alex@alex:~/Micro-XRCE-DDS-Gen$ cd
alex@alex:~$ cd ardu_ws/
alex@alex:~/ardu_ws$ colcon build
Starting >>> ardupilot_msgs
Starting >>> micro_ros_agent
Finished <<< micro_ros_agent [0.13s]
Finished <<< ardupilot_msgs [0.44s]
Starting >>> ardupilot_sitl
[Processing: ardupilot_sitl]
[Processing: ardupilot_sitl]
[1min 17.5s] [2/4 complete] [ardupilot_sitl:build 100% - 1min 16.7s]
```